robotemi/sdk #9: turnBy calls

padresb, 2019-09-01:
When calling turnBy(90,6.2), not getting any movement.  I have been playing with skidJoy and calls to this function do move Temi.

Here is the log entry:

09-01 16:12:17.884 9741-9741/com.robotemi.sdk.sample.china D/Robot: turnBy(int, float) (azimuth=90, speed=6.2)

padresb, 2019-09-01:
tried speed = 1.0 and still no movement

ramisaban1, 2019-09-04:
Hello,

I hope all is well. Please change your **Build Variant** to install the USA version not the Chinese one. 

This can happen due to multiple reasons. Just so I understand better:

1) Is your robot moving when you tap on the sample app's skidJoy button?
2) Are you running the latest changes on the master branch of the sample app? If not which branch are you running?

Keep in mind:
1) To use correct variant of the SDK (i.e. USA one vs. Chinese one when necessary).
2) That an object is not blocking the path the robot would turn to.

closed 2019-09-28 by ramisaban1