robotemi/sdk #18: Temi doesn't respond to Sample App

QuantJames, 2019-10-22:
After pushing `sample` APK via Android Studio/ADB, the sample app loads successfully, and onClick events are registered, but the robot does not otherwise move or respond at all. We're not sure how to debug this, any assistance would be greatly appreciated! We tried building and pushing from two devices (Mac and Windows)

@ramisaban1 

```D/Robot: tiltBy(int) (degrees=70)
D/Robot: turnBy(int) (degrees=23)
D/Robot: tiltBy(int) (degrees=70)
I/InputMethodManager: lhk  showSoftInput()send soft input window show message !!!!!!!!!!
D/Robot: tiltBy(int) (degrees=70)
D/Robot: toggleWakeup() - disable = true
D/Robot: tiltBy(int) (degrees=70)

QuantJames, 2019-10-22:
Addendum... we also get exceptions when certain buttons are pressed, e.g.

```10-22 12:49:27.061 6611-6611/com.robotemi.sdk.sample.china E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.robotemi.sdk.sample.china, PID: 6611
    java.lang.IllegalStateException: Could not execute method for android:onClick
        at androidx.appcompat.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:402)
        at android.view.View.performClick(View.java:5204)
        at android.view.View$PerformClick.run(View.java:21177)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662)
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at androidx.appcompat.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:397)
        at android.view.View.performClick(View.java:5204) 
        at android.view.View$PerformClick.run(View.java:21177) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:148) 
        at android.app.ActivityThread.main(ActivityThread.java:5417) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662) 
     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.robotemi.sdk.BatteryData.isCharging()' on a null object reference
        at com.robotemi.sdk.sample.MainActivity.getBatteryData(MainActivity.java:250)
        at java.lang.reflect.Method.invoke(Native Method) 
        at androidx.appcompat.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:397) 
        at android.view.View.performClick(View.java:5204) 
        at android.view.View$PerformClick.run(View.java:21177) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:148) 
        at android.app.ActivityThread.main(ActivityThread.java:5417) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662)

bolatashim, 2019-10-24:
The latest version of the repository seems to have certain issues with regard to Robot control. I also faced this problem myself. Try using this snapshot https://github.com/robotemi/sdk/tree/e1e1c674446f247d70cf7a7c4a3e3a4cb94c19cf . It has pretty much all of the features that are present in the last commit and has no problems controlling the robot hardware.

QuantJames, 2019-10-25:
Thanks, seems I was using the wrong build variant.

closed 2019-10-25 by QuantJames