robotemi/sdk #27: Detect User on Welcoming Mode

bruno963852, 2019-11-19:
Is there some way for the application to know when the robot greets a user on welcoming mode? I want to program the robot to say something or to show some greeting screen.

bruno963852, 2019-11-19:
I found the answer in the documentation, but there's another problem. There's no addOnWelcomingModeStatusChangedListener() methon on the Robot object.

ramisaban1, 2019-11-19:
Hello,

Using Welcome mode within the SDK including addOnWelcomeModeStatusChangedListener will be available on temi’s next Software Update. This has been much requested and will be available in the next coming weeks.

Best,
Rami

closed 2019-11-25 by ramisaban1

bruno963852, 2019-12-18:
Any news on this matter? It looks like it is still not implemented.

ramisaban1, 2019-12-18:
Hello @bruno963852,

In the next temi Software Update, we will be releasing an improved SDK containing the exact methods that you are looking for.

If you are looking for a reference for right now, feel free to check out our B2B welcome app at https://github.com/ROBOTEAM-HOME/temi-welcomingmode-b2b