robotemi/sdk #29: Get SLAM map?

uglyduck2019, 2019-11-21:
I have 2 questions...
1. I'd like to get SLAM map and show this on my APP, how should I do? (also position of current Temi, points of locations been saved)
2. Is that possible to set location on map directly without guild Temi to the location then saved?
thanks for any answers

lcarnevale, 2019-11-27:
Hello everyone,
I am also interested to the @uglyduck2019 questions.
I follow this thread.

ramisaban1, 2019-12-03:
Hello all,

Thanks for reaching out and sorry for the delay- Currently, access is only available to the tablet portion of the robot. You can do things like, `Robot.getinstance().goto("kitchen")` which sends commands from the tablet to the Linux-running computer for navigation. You can do this within an android 6.0 compatible application located on the tablet of the robot.

Mostly, we don't allow access to the Linux-running computer at temi's base for security purposes. This computer is in charge of all navigation components including path planning, obstacle detection, follow mode and much more.

Sorry I couldn't be of more help. Let us know if you need anything else!

uglyduck2019, 2019-12-04:
Hi,
Thanks for your reply.
Now, we can see the map tap the item "MAP" in setting. How do I implement the MAP to real-time displayed on my APP?

BR,

ramisaban1, 2019-12-04:
Currently, the only place you can get the live feed of the map is in Settings -> Map. Receiving this exact live image of the map is listed as a request for SDK development but is not yet implemented for public use.

closed 2019-12-04 by ramisaban1