robotemi/sdk #30: How many instances are running?

mediadogg, 2019-11-23:
I am just beginning to design, build and test my first temi app, which is also my first Java app, and my first Android app. So, please be gentle!

I am confused about how many instances of my app are running. Each time I launch it from the Home screen, I get a message that I create within the default constructor, which is where I also start a background task for communications with my PC app. I can still communicate the PC when other temi apps (such as Youtube) are running. This is wonderful. But does that mean that I have have been creating multiple instances of my background app (a new one created each time I launch my app?) unless I manually control it?

abduelhamit, 2019-11-25:
Your question is about Android, not temi-specific. As long as your question is not about the stuff implemented in this SDK, assume that it's about Android.

Only one instance of your app will run on Android, but there can be multiple [activities](https://developer.android.com/guide/components/activities/intro-activities). I don't precisely know what you want to do, but I think that [`onCreate()`](https://developer.android.com/guide/components/activities/intro-activities#oncreate) is the right place.

I recommend you to read [the guide to app architecture](https://developer.android.com/jetpack/docs/guide) and learn more about [the activity lifecycle](https://developer.android.com/guide/components/activities/activity-lifecycle).

lcgao, 2019-11-25:
Check the launch mode of your MainActivity

mediadogg, 2019-11-25:
Thanks for the replies and the references to helpful information. Everybody starts somewhere, and I have just begun my journey in this domain, although being quite experienced in others. And nobody knows it all, or writes bug-free code. If I knew enough to determine precisely what was temi- specific. I would not have needed to ask the question, and would not need this forum.

Once I do gain the skills I seek, I will definitely remember from whence I started and try to avoid talking down to newbies.

ramisaban1, 2019-11-25:
Looks like the issue here was solved.

Feel free to reopen it if needed.

closed 2019-11-25 by ramisaban1