robotemi/sdk #31: Speech-To-text after WakeupWord trigger

lcarnevale, 2019-11-27:
Hello all,

This issue partially covers the topic discussed into the issue #24, which is currently closed.

I have tried the sample application into this SDK and I have understood how to use the _WakeupWordListener_ to activate the Speech Recognition of Temi. Therefore, I followed the tutorial posted into the issue #24 to setup an intent through the Temi Settings page and implement my code into _onNlpCompleted_. This works pretty fine.

What I have wondered is why this tutorial involves a manual process and therefore I have few questions:
1. may I use a programmatic approach to setup the intents?
2. may I access the Speech-To-Text, which is available into _onNlpCompleted_, just after the wake up? I mean, I would like to have a channel which transcribes what the user says to the robot after the wake up. Is an API call available for accomplish this goal?

Best,
Lorenzo.

ramisaban1, 2019-12-04:
Hello,

Thank you for your concern, the Local NLP functionality was always intended as something temporary. This was implemented in order to give developers something to work with until February which is when we plan to allow usage of a custom DialogFlow NLP Agent.

Best,
Rami

ramisaban1, 2019-12-16:
Let me know if you have any additional questions or concerns

closed 2019-12-16 by ramisaban1

PureSecDE, 2020-03-12:
> This was implemented in order to give developers something to work with until February which is when we plan to allow usage of a custom DialogFlow NLP Agent.

Hello @ramisaban1 ,

is it still planned to make it possible to use an own DialogFlow-Agent and is there already an approximate date when you can expect this?

Best,
PureSecDE

nicolacorriero, 2020-04-08:
> Hello,
> 
> Thank you for your concern, the Local NLP functionality was always intended as something temporary. This was implemented in order to give developers something to work with until February which is when we plan to allow usage of a custom DialogFlow NLP Agent.
> 
> At this point in time, there is no programmatic approach to setting up intents. Since we are working on releasing this to the public as quickly as possible, we are focussing mainly on its' development, and not improving the current Local NLP.
> 
> Regarding a channel to get what the user says to the robot after the wake-up, you can use `NlpResult.resolvedQuery` within the `onNlpCompleted` listener which returns a `string` containing the command given by the user.
> 
> Best,
> Rami

Hello @ramisaban1 
any news regards custom DialogFlow NLP Agent?

nicolacorriero, 2020-04-08:
Hello @ramisaban1 
any news regards custom DialogFlow NLP Agent?