robotemi/sdk #33: Cancel a TtsRequest

lcarnevale, 2019-11-27:
Hello all,

I am looking for an API call that cancel a running TtsRequest, like the following one:
```
fun speak() {
  val ttsRequest = TtsRequest.create("Hello, this is a long sentence", false)
  robot.speak(ttsRequest)
}
```

Is this one available?

Best,
Lorenzo.

ramisaban1, 2019-12-03:
You can cancel a running TtsRequest by calling the `Robot.getInstance().cancelAllTtsRequests()` method.

Please let me know if this solved your question.

lcarnevale, 2019-12-04:
Hello @ramisaban1 ,

I successfully tried out the _cancelAllTtsRequests_ method.

In addition, I have found out another solution at this issue. Even if it is not the proper way, I would like to share it here.
While a TtsRequest is raised up, another one with a white space (" ") executes a silent speech, overwriting the existing one.

```
fun speak() {
  val ttsRequest = TtsRequest.create(" ", false)
  robot.speak(ttsRequest)
}
```

Best,
Lorenzo.

ramisaban1, 2019-12-04:
Thank you, Lorenzo.

Best,
Rami

closed 2019-12-04 by ramisaban1