robotemi/sdk #38: Support for emulator

kkumpavat, 2019-12-09:
Hi All,

I will be starting work on Temi robot app and the Initial development is only related to UI. So, I was just wondering if we have emulator support to test the UI part? If not, does UI developed for normal android tablets work well on Temi Tablet? Also, I wanted to know any restrictions put on custom android OS.

Thanks,
Kuldeep

hermanShai, 2019-12-09:
Hi Kuldeep,

Thanks for reaching out.
We don't have a temi dedicated emulator, if you are just developing UI you can use a 10 inch 1280*1800 tablet emulator that should be good enough.

I'm not sure exactly what you are asking here? 
> I wanted to know any restrictions put on custom android OS.

kkumpavat, 2019-12-10:
> Hi Kuldeep,
> 
> Thanks for reaching out.
> We don't have a temi dedicated emulator, if you are just developing UI you can use a 10 inch 1280*1800 tablet emulator that should be good enough.
> 
> I'm not sure exactly what you are asking here?
> 
> > I wanted to know any restrictions put on custom android OS.

Temi robot OS is based on Android OS 6.0.1. I wanted to know if all the 6.0.1 capabilities are present in TEMI OS as well or are there any restrictions?

hermanShai, 2019-12-12:
There shouldn't be any restrictions, please let us know if you do encounter any.

ramisaban1, 2019-12-16:
Given that we answered this, I'm going to close out the issue, but feel free to get back in touch if you're still having issues!

closed 2019-12-16 by ramisaban1