robotemi/sdk #40: Temi ADB Connection Issue on temi version 115.64

muhadnanafridi, 2019-12-16:
I tries severial networks to connect my android studio with temi  but it is giving me this error on command "adb connect 172.18.20.152:5555"

"cannot connect to 172.18.20.152:5555: No connection could be made because the target machine actively refused it. (10061)"

ramisaban1, 2019-12-16:
Hello @muhadnanafridi,

Most of the time this type of thing happens if **ADB Port Opening** wasn't pressed under **Settings -> _temi_ Developer Tools -> ADB Port Opening**. If this still doesn't work, please make sure both the computer you are trying to connect from as well as _temi_ are connected to the same Wi-Fi network.

Best,
Rami

datablackhole, 2019-12-17:
Same issue faced by me. I have tried to connect via adb using multiple computers and also tried different networks where i made sure I both the robot and computer were on the same network...

I've noticed Temi technician tools menu which asks for a password... is there anything in that menu that could be of help ?

lcarnevale, 2019-12-19:
Hello,

I experienced a closing of the ADB Port after a while. Therefore, even if I already opened it and connected, I had to open again.

Best,
Lorenzo.

ramisaban1, 2019-12-19:
Are you tapping **ADB Port Opening** after connecting to the network on temi? The only reason the "device" (meaning temi) would refuse a connection is if the port wasn't opened.

datablackhole, 2019-12-24:
> Are you tapping **ADB Port Opening** after connecting to the network on temi? The only reason the "device" (meaning temi) would refuse a connection is if the port wasn't opened.

Several times.

muhadnanafridi, 2019-12-24:
Yes I am tapping ADB PORT Opening but its not working for me . From last 1 week i am unable to work on Temi .

ramisaban1, 2019-12-26:
Hello,

I know this seems frustrating, and you definitely tried this, but please try connecting both devices to a new network. I am also attaching a video demonstrating the connection process.

ADB Connecting Tutorial Video:
https://drive.google.com/file/d/1NaQda3udF-LrXmJ6pisSwDZ2XUCD_2d5/view?usp=sharing

Best,
Rami

uglyduck2019, 2019-12-27:
We had two Temi on hand, and one has this problem that cannot connect with ADB tools. We had tired everything we can. After we did "FACTORY RESET", Temi is dead that cannot go to home page (lock on screen "waiting for docking" (initial procedure))

ramisaban1, 2019-12-27:
Hello @uglyduck2019,

Please reach out to our support team via the mobile app or the website for further troubleshooting. This issue does not have anything to do with opening the ADB port.

datablackhole, 2019-12-28:
> home
 Hey there uglyduck2019,

We have same exact problem now. We can't get past docking screen after factory reset. The support team said to tap on the top left corner of the display to get past this docking screen but now it just goes back to the same screen again and again.

Dear ramisaban1,
We have done exactly what you've showed in the video. No luck. But we are in contact with temi support team at moment. And no luck so far.  Is there a way to downgrade the software ?! because right now its nearly impossible to even that since we have no way to connect to temi i.e. via wifi. Is there a hidden usb port ?! .

datablackhole, 2019-12-28:
> Hello @uglyduck2019,
> 
> Please reach out to our support team via the mobile app or the website for further troubleshooting. This issue does not have anything to do with opening the ADB port.

Dear ramisaban1,
This issue does **HAVE** a thing to do with opening the ADB port. Because this very ADB issue lead us to factory reset the device after having it upgraded to the latest software update.

muhadnanafridi, 2019-12-29:
@ramisaban1 I tried to connect with 5 different networds  but its still not working .

ramisaban1, 2019-12-31:
If anyone active on this issue is experiencing this and available to troubleshoot (with multiple networks, preferably a mobile hotspot), please reach out to live chat and ask for Rami. I should take over the chat and work with you from there.

ramisaban1, 2020-01-16:
If anyone is still having this issue, please reach out to our live chat either on our website or directly from the temi mobile app on your smartphone and ask for Rami.

datablackhole, 2020-02-20:
Temi team sent replacement robot because of this issue.

ramisaban1, 2020-02-20:
Hello @datablackhole ,

Please reach out to live chat today and ask for Rami. I would love to speak to you regarding that issue.

ramisaban1, 2020-02-20:
Otherwise, when would be a good time to speak?

leslieam, 2020-06-08:
Please try with anther network. For example, your hotspot.

You network may have some policies which blocked the connection.

closed 2020-09-18 by lcgao

nteku, 2023-01-09:
Hello! I am trying to connect my Temi Robot to a WPA-Enterprise network, but it is not letting me. Does anyone know how I can fix this?