robotemi/sdk #47: HideTopBar in Kiosk Mode

leeykdydrb, 2020-01-14:
When you apply hideTopBar in kiosk mode, the location did not appear at the top of the screen in the past, but now appears.

I want to not show.

And if you apply hideTopBar in kiosk mode, calling hey temi doesn't work, but now it reacts.

I wish no reacts.

What's wrong with the Temi software update?

abduelhamit, 2020-01-15:
At least you can disable voice wakeup with `toggleWakeup(true)`.

ramisaban1, 2020-01-16:
Hello @leeykdydrb ,

The Green Bar (which included messages like 1. Searching 2. Following 3. ****** Obstacle and 4. LocationName) displayed was separated from the top bar by design because technically it shouldn't be part of the top bar. Toggling whether or not to show the Green Bar is listed as an SDK request and will be prioritized according to severity. However, I don't have a timeline as to when this will be implemented.

Feel free to reach back out here or at www.robotemi.com/developer-support for faster responses.

Best,
Rami

closed 2020-01-16 by ramisaban1