robotemi/sdk #50: temi charging

Lee827, 2020-02-06:
What is the lowest battery energy temi will return back to the home base? Because I want to control return back home base in my apps. Can I close the auto-return back? (For charging)

ramisaban1, 2020-02-10:
Hello,

Thank you for reaching out to temi's SDK team. You can easily call our Battery Listener to listen for any percent battery you want to return to the Home Base at and send the robot there.

Feel free to check out our documentation [here](https://github.com/robotemi/sdk/wiki).

Hint: Just add the Battery listener just like you would use any of temi's other listeners. 

Best,
Rami

closed 2020-02-10 by ramisaban1

Lee827, 2020-02-11:
I know the Battery Listener. I am afraid your auto-return back function will run first. So I want to know the lowest battery energy and close the function. Let me easily to handle.