robotemi/sdk #63: fetch map from ROBOX

uglyduck2019, 2020-03-12:
Hi,
as the release note version #118
"Adding API support for map fetch and goto coordinates"
Where I can find the API and relate document or code?
thanks

Vorlif, 2020-03-12:
Hi all,

where can I find the release notes?

PureSecDE, 2020-03-12:
Hi, everybody,

where can I find the release notes?

chris-lsn, 2020-04-16:
Do you guys have any update about it?

Thiago-LM, 2020-04-23:
It's strange 'cause since [this](https://github.com/robotemi/sdk/commit/f29c12dc03a56543f5dda82a977706c75595ccb7) commit, from Jan 20, that's already a map function, but only on branch [map2](https://github.com/robotemi/sdk/tree/map2) and [mapMerge](https://github.com/robotemi/sdk/tree/mapMerge). And I don't know how to use them on my Android Studio project.

cyrsis, 2020-06-04:
it would be great to have this working.

So it is the indoor map so that I can just use it a reference for the indoor nav system?

leslieam, 2020-06-08:
We are working on this. It will be ready in a few versions ahead.

closed 2020-06-08 by lcgao