robotemi/sdk #64: Toggle Welcoming Mode

brad-oosterveld, 2020-03-12:
Is it possible to toggle welcoming mode on/off? It would be great to be able to have the robot only search for people at certain times.

From what I can tell, I have complete control over custom reactions to detection/interaction. I can write my code to only react in certain circumstances, but I have not figured out how to prevent the robot from "locking on" to the person that it detects and rotating to follow them.

Would it be possible to have something like constraintBeWithMe() or beWithMe() send messages to the detection listener even if the robot isn't in welcoming mode?

Thanks!

brad-oosterveld, 2020-03-13:
It seems like most of the behavior I want can be achieved by using `beWithMe()` and the associated listener and tehn immediately calling `stopMovement()` after detection occurs. This can be done when the robot is not it welcoming mode and does not require use of the detection/interaction listeners.

I am not sure if there is a way to stop `beWithMe()` other than calling `stopMovement() `or `goTo()`. It is not apparent from the documentation that `stopMovement()` also stops person detection, but empirically it seems to be the case.

closed 2020-03-13 by brad-oosterveld

stawberryslayer, 2024-12-16:
Thank you for the solution, I have been also looking foe ways to disable `beWithMe()`