robotemi/sdk #65: Incorrect rotation with screen sharing

PureSecDE, 2020-03-16:
Hello, all, everyone,

we want to use the Android screen sharing function with Temi but unfortunately the screen is always rotated by 90 degrees. Is there a way to fix this problem? It seems so, so the tablet would think it's displayed in portrait mode.

To reproduce the problem easily you can also use the program [scrcpy](https://github.com/Genymobile/scrcpy), here we have the same failure and think that it is related.

Best,
PureSecDE

[image]

leslieam, 2020-06-08:
It will be fixed in a new firmware.

icnocop, 2020-08-03:
Hi.

`adb shell wm size` returns `Physical size: 1200x1920` which is incorrect.

You can pass `--rotation 3` to `scrcpy.exe` and the screen will be rotated 90 degrees clockwise, but it will still be cut off.

See https://github.com/Genymobile/scrcpy#rotation

`The rotation can also be changed dynamically with Ctrl + ← (left) and Ctrl + → (right).`

I tried to call `adb shell wm size 1920x1200` and `adb shell wm overscan ...` but I wasn't able to fix the issue.

leslieam, 2020-08-03:
Thanks for the info.
We've fixed the rotate issue in the new firmware and will release after we finish the test.

hehao85, 2020-08-15:
How do I get the new firmware ?

leslieam, 2020-08-17:
We'll release it when it is ready. You Should receive an update on your robot.

closed 2020-08-17 by leslieam

dimitriosp, 2020-11-23:
Hello, reopening this one since it seems that it is still rotated and cut off. 
[image]
any update on this? 
best