robotemi/sdk #89: Completely hide the top bar.

Thiago-LM, 2020-06-08:
Is there any way to completely hide the top bar? I mean, remove even the little white thing and the ability to "pull down" the top bar.

At the moment it can only get to this:
[image]

And if there isn't any way at the moment, is it planned to at some point? There are other methods I could use to circumvent this?

I want to lock my application at temi's head and only let someone authorized exit it.

lcgao, 2020-06-22:
1. Add this `meta-data` to specific `Activity` flag in **AndroidManifest.xml**.

    ``` xml
    <meta-data
        android:name="com.robotemi.sdk.metadata.UI_MODE"
        android:value="4" />
    ```
    For example,

    ``` xml
    <activity
        android:name=".MainActivity"
        android:launchMode="singleTask">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>

            <meta-data
                android:name="com.robotemi.sdk.metadata.UI_MODE"
                android:value="4" />
    </activity>
    ```

2. Add following method to your `Activity` class and call it.

    ``` java
    private void refreshTemiUi() {
        try {
            ActivityInfo activityInfo = getPackageManager()
                    .getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
            Robot.getInstance().onStart(activityInfo);
        } catch (PackageManager.NameNotFoundException e) {
            e.printStackTrace();
        }
    }
    ```

Thiago-LM, 2020-06-22:
Thanks for the response, but I can't see to get this to work. Where should I call `refreshTemiUi()`? On my onCreate? onStart? I've tried both and it didn't work. Also, should I remove `hideTopBar()` from my code and use this only instead?

lcgao, 2020-06-23:
Hi, please consider calling `refreshTemiUi` in the callback method `onRobotReady(boolean isReady)` and only when `isReady == true`. No need to use `hideTopBar()` if you used `refreshTemiUi` instead.

Thiago-LM, 2020-06-23:
It is working now. Thank you.

closed 2020-06-23 by Thiago-LM

prashantsmp, 2022-11-17:
> Hi, please consider calling `refreshTemiUi` in the callback method `onRobotReady(boolean isReady)` and only when `isReady == true`. No need to use `hideTopBar()` if you used `refreshTemiUi` instead.

How to exit from this application after removing top bar? There is no option to go back to the Main menu?

zjn0505, 2022-11-18:
There is a `startPage` method in SDK.

oec-himal, 2024-12-25:
@lcgao and others, 
I have got one problem. 
I have implemented according to the above technique and it worked but, 
top bar reappears when I return back from other page by following code:

backButton.setOnClickListener{
            onBackPressedDispatcher.onBackPressed()
        }

This above code transitions to the MainActivity but, Top bar Reappears.
I tried by calling refreshTemiUi() from on Resume lifecycle hook but, doesn't work.