robotemi/sdk #90: Support testing apps in Android emulator

WasabiFan, 2020-06-09:
As mentioned in #84, the Temi SDK kills the process if the Temi service isn't present on the device. This makes it very difficult to quickly test applications. Can you change this logic so that it allows the app to continue running if not on Temi hardware and/or if it's in the emulator? Alternately, could you add an option to avoid trying to initialize the Temi connection if I tell the SDK not to?

For others, the relevant code is here: https://github.com/robotemi/sdk/blob/0b3bf9c604f55e7286b6107111ec121b28d8737d/sdk/src/main/java/com/robotemi/sdk/TemiSdkServiceConnection.java#L65

lcgao, 2020-07-13:
Hello, temi SDK is open-source already, so developers can remove `android.os.Process.killProcess(android.os.Process.myPid());` and **re-build** the source code to a **.aar** file(**sdk/build/outputs/aar/sdk-debug.aar**), and your project can use this **.aar** file as the temi SDK dependency when you just want to test the Apps in other Android devices like the emulator. But please use the **official** temi SDK on Maven Central when the Apps run on temi.

WasabiFan, 2020-07-14:
@lcgao Unfortunately, I don't think custom builds are a viable solution here.

This would require any/all developers on a project to make a custom build or share it between them, and then edit the Gradle config and project settings each time they wanted to switch between emulator and Temi. That's a large amount of extra work and gets in the way of source control.

Since you can be sure that the Temi APIs will not work when not on the Temi, it seems it would make more sense to check the device and only attempt connection (or only kill the app) when on Temi. It is a simple addition, and would work for all users rather than requiring each person to fix it themselves and change their project.

bn-dignitas, 2020-08-04:
@lcgao What is the intent of putting the killProcess in the code?  Even though a dev could recompile after commenting this out, will they run into other issues later?  If it is not necessary, and just causes trouble, why include it?

The issue as @WasabiFan alludes is developer convenience and the reality that most teams probably do not have a temi for every developer.  Allowing developers to test their non-temi dependent code before putting it on the hardware is a huge bonus.  When weighing platforms to use, ease of development is often a key factor when choosing between them.  These kinds of changes, if not necessary, will not help your platform long term.

dimitriosp, 2020-09-17:
Hello, 
any update on this since we would also need the emulator functionalities.

handicanam, 2021-11-24:
@lcgao Our company is in the process of wanting to evaluate Temi robots.
I hope that the management team will strongly consider again on what @WasabiFan mentioned.
This will greatly help both the devs and your company.
Thank you

dimitriosp, 2022-02-17:
digging this out again. any update?