robotemi/sdk #91: 如何禁止temi的NLP或如何添加我们的语音库到temi

laijeffery, 2020-06-17:
Hi,

我们这边有个需求，就是我们之前本地开发了一些我们行业的一些语音库放在我们的后台server,
现在当我们的APP在前台运行，我们本来用asr(onAsrResult)获取用户说的语音结果后，传往后台，但是temi会自动理解、搜索客户说的话，并作出回答， 而实际上我们后台server的语音回答更专业准确，但是被temi先自动回答了。

我们的问题时，能否禁止temi的自动语音处理功能，我们只用asr获取用户说的话，拿到后台，得到我们的专业答案后，用tts来播报给用户， 不然temi会去理解播报一下，这是我们所不想要的。

此外，如果temi不能在我们的APP中禁止自然语言处理，那么能否把我们的语音、语料库，加到你们temi里面去呢，如果可以，怎么操作呢？

Thanks

lcgao, 2020-06-17:
Hi,
119版本的temi+0.10.6 temi SDK可以选择完全覆盖temi原有的NLP流，只接收ASR结果，可以考虑在onAsrResut()中接入开发者自己的NLP服务。参考如下：
https://github.com/robotemi/sdk/wiki/Speech_chn#addAsrListener
https://github.com/robotemi/sdk/blob/0b3bf9c604f55e7286b6107111ec121b28d8737d/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L539-L576

laijeffery, 2020-06-17:
Hi 
是的，我们有注意到设置里面本地NLP这个功能，但是，我们有个问题，我们语音太多了(>1000条)， 而且，有个模糊匹配的问题，比如，“”办业务“”， “”我要办业务“”， “”请帮我办业务“”， 这些都是类似的句子，我都要一个个输入到设置里面本地NLP，添加Action吗，那样太多太多了，没法手动输入几千条， 加上模糊匹配问题，我们，不可能把客户说的话的各种表达方式，都加到本地NLP

所以才提出以上问题，烦请协助

Thanks

laijeffery, 2020-06-17:
@lcgao

lcgao, 2020-06-17:
> Hi
> 是的，我们有注意到设置里面本地NLP这个功能，但是，我们有个问题，我们语音太多了(>1000条)， 而且，有个模糊匹配的问题，比如，“”办业务“”， “”我要办业务“”， “”请帮我办业务“”， 这些都是类似的句子，我都要一个个输入到设置里面本地NLP，添加Action吗，那样太多太多了，没法手动输入几千条， 加上模糊匹配问题，我们，不可能把客户说的话的各种表达方式，都加到本地NLP
> 
> 所以才提出以上问题，烦请协助
> 
> Thanks

Hi，我指的不是本地NLP，就是新版本才有的覆盖原有语音流NLP，通过声明我刚才的发的`meta-data`就可以从ASR结束后切断temi原有的流程，从而转到你们自己的NLP服务。

laijeffery, 2020-06-17:
Hi,

我再确认下，你说新版条件是，机器人119， sdk 10.63?

另外，您说的操作方法是这样吗，

Step1
manifest声明：
      * <meta-data 
      *     android:name="com.robotemi.sdk.metadata.KIOSK" 
      *     android:value="TRUE" /> 
      * 
      * <meta-data 
      *     android:name="com.robotemi.sdk.metadata.OVERRIDE_NLU" 
      *     android:value="TRUE" /> 

Step2
同时， 把我们的APP，设置为Kiosk Mode：
And also need to select this App as the Kiosk Mode App in Settings > Kiosk Mode. 

这时，到asr收到客户语音后， temi，就不会去理解、搜索并响应客户说的话？
也就是说，假设，这个时候，我不什么都不做， temi就只有asr接收语音，以及手动触发tts发出语音的功能？

Thanks

lcgao, 2020-06-17:
> Hi,
>
> 我再确认下，你说新版条件是，机器人119， sdk 10.63?
>
> 另外，您说的操作方法是这样吗，
>
> Step1
> manifest声明：
> * <meta-data
> * android:name="com.robotemi.sdk.metadata.KIOSK"
> * android:value="TRUE" />
> *
> * <meta-data
> * [...]

对，temi原有流程只会到ASR环节。

laijeffery, 2020-06-17:
好的，我们明天试下，有问题再请教您， thanks

laijeffery, 2020-06-19:
@lcgao 
Hi 
前两天的问题解决了。

现在有另一个问题：

public void onGoToLocationStatusChanged(@NotNull String location, String status, int descriptionId, @NotNull String description) {
        Log.d("GoToStatusChanged", "status=" + status + ", descriptionId=" + descriptionId + ", description=" + description);
        robot.speak(TtsRequest.create(description, false));
        switch (status) {
            case "start":
                robot.speak(TtsRequest.create("Starting", false));
                break;

            case "calculating":
                robot.speak(TtsRequest.create("Calculating", false));
                break;

            case "going":
                robot.speak(TtsRequest.create("Going", false));
                break;

            case "complete":
                robot.speak(TtsRequest.create("Completed", false));
                break;

            case "abort":
                robot.speak(TtsRequest.create("Cancelled", false));
                break;
        }
    }
机器人在导航以及行走中不能说话，onGoToLocationStatusChanged是有回调到，robot.speak，不能说话，请帮忙分析下，感谢

laijeffery, 2020-06-19:
@lcgao

jchen932, 2022-09-28:
.