robotemi/sdk #92: Add dynamic location points from api

dimitriosp, 2020-06-17:
Hello,
one of our customers asked if its possible to dynamically add location points to the map, without having temi physically drive there. 
This would need to get the map and be able to add new locations and update the map on the robot.

best 
Dimitrios

lcgao, 2020-06-17:
Hello, We are working on this, but this is only for the Web platform not for SDK.

dimitriosp, 2020-06-17:
So the customer app could add dynamic locations to the cloud, but how does the robot receive the updated map then?

lcgao, 2020-06-18:
Customer's App couldn't add dynamic locations to the cloud directly. By the way I want to the question you asked that did you just want to add a location(location name) in your own App? if so, you could use `Robot.getInstance().saveLocation(locationName)` to implement that and no need to get the map.

closed 2020-09-04 by lcgao

dimitriosp, 2020-09-17:
Hello, reopening this, sorry for the delay.

The use case would be:
- map of the environment is available on the robot and cloud
- a new dynamic previously unknown location is added to the cloud map
- the map is pushed to temi and loaded as new map
- now the robot navigates to the new location

Is this possible?
best Dimitrios