robotemi/sdk #95: robot can't speak during moving

laijeffery, 2020-06-22:
Hi
前两天的问题(issue#91)解决了。

现在有另一个问题：

public void onGoToLocationStatusChanged(@NotNull String location, String status, int descriptionId, @NotNull String description) {
Log.d("GoToStatusChanged", "status=" + status + ", descriptionId=" + descriptionId + ", description=" + description);
robot.speak(TtsRequest.create(description, false));
switch (status) {
case "start":
robot.speak(TtsRequest.create("Starting", false));
break;

        case "calculating":
            robot.speak(TtsRequest.create("Calculating", false));
            break;

        case "going":
            robot.speak(TtsRequest.create("Going", false));
            break;

        case "complete":
            robot.speak(TtsRequest.create("Completed", false));
            break;

        case "abort":
            robot.speak(TtsRequest.create("Cancelled", false));
            break;
    }
}
机器人在导航以及行走中不能说话，onGoToLocationStatusChanged是有回调到，robot.speak，不能说话，请帮忙分析下，感谢

lcgao, 2020-06-22:
`Robot.getInstance().speak(TtsRequest ttsRequest)` 需要连接互联网，检查一下网络状况。另外可以监听一下 `TtsListener` 看是否有具体的状态回调提示。

laijeffery, 2020-06-22:
Thanks for your help, we  have fixed it.

closed 2020-06-22 by lcgao