robotemi/sdk #98: Possible to let temi say something when reached to some location?

MeiShin00, 2020-07-14:
How to let temi say something when temi reached and  stop at the location?

lcgao, 2020-07-16:
Hello, please try to listen to the [OnGoToLocationStatusChangedListener](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener) to get the status of  **Go To Location**, so you can invoke `Robot.getInstance().speak(TtsRequest.create("Arrived xxx location", false))` when the status is `COMPLETE` or `ABORT`. For more details please refer to the [Sample code](https://github.com/robotemi/sdk/tree/master/sample).

Implement the listener interface and override the method:
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L64
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L463-L488
Add listener:
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L117
Remove listener:
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L137

MeiShin00, 2020-07-17:
Hi, I solve my problem, thanks for your help and your reply !

closed 2020-07-17 by lcgao