robotemi/sdk #100: Builtin camera crashes whenever I take a photo

LuisFernandezAlisys, 2020-07-23:
Hello, I'm developing an app for temi and whenever I take a picture with ACTION_IMAGE_CAPTURE the intent result is false and the camera app (not my app, the builtin android camera) crashes and shows an error dialog with the message "Camera error: Can't connect to the camera"

In a regular device, clearing the cache of the camera app might help but since we don't have access to that opcion on temi, what would be a valid solution for this?

lcgao, 2020-07-24:
Generally this issue can be solved after rebooting temi.

LuisFernandezAlisys, 2020-07-24:
I've tried rebooting the robot and it still doesn't work. I tried in a couple different robots just in case and it still gives the same error.

icnocop, 2020-07-30:
From https://developer.android.com/training/camera/photobasics#TaskCaptureIntent

> Notice that the startActivityForResult() method is protected by a condition that calls resolveActivity(), which returns the first activity component that can handle the intent. Performing this check is important because if you call startActivityForResult() using an intent that no app can handle, your app will crash. So as long as the result is not null, it's safe to use the intent.

closed 2020-07-31 by LuisFernandezAlisys