robotemi/sdk #103: Interupt with "userDetection" while "goToLocation"

vicnando, 2020-07-29:
Hi , i have a problem with user detection. In my case i need to make my temi to patrol and detect people nearby and say greetings. But when i use robot.goTo("location") . The detection Listener is not trigger at all.

Is that because if temi focus on go to location , all listener will be delayed until Temi arrive to somewhere location ?

Please help me, Thanks!

lcgao, 2020-07-31:
Yes, that's correct, detection mode will stop when temi focus on go to location.

vicnando, 2020-07-31:
Btw. Can we make the detection more wider and far.  Is there any idea for
that? Thx for your response

lcgao, 2020-07-31:
Developers can extend the detection distance in next offical version.

vicnando, 2020-07-31:
So, can i know when the the next version will be release then?

lcgao, 2020-08-03:
It's still in testing, and updates will be pushed when it is officially released.

closed 2020-09-18 by lcgao