robotemi/sdk #108: How to add application button to top bar?

icnocop, 2020-08-04:
Hi.

I'm wondering if it's possible to add my application's icon to the top bar?

I see that after some applications are launched, an icon will be added there.

[image]

Thank you! :)

ray-hrst, 2020-08-04:
@icnocop those instructions are provided in the [wiki](https://github.com/robotemi/sdk/wiki) (end of the page):

```
@Override
public void onRobotReady(boolean isReady) {
    if (isReady) {
        try {
            final ActivityInfo activityInfo = getPackageManager().getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
            robot.onStart(activityInfo);
        } catch (PackageManager.NameNotFoundException e) {
            throw new RuntimeException(e);
        }
    }
}
```

icnocop, 2020-08-04:
Thank you!

I had to do the following:

1. Implement `OnRobotReadyListener`
2. Declare variable `private lateinit var robot : Robot`
3. Initialize variable:
```
override fun onCreate(savedInstanceState: Bundle?) {
        ...
        this.robot = Robot.getInstance(); // get an instance of the robot in order to begin using its features.
        ...
    }
```

4. Add override functions:
```
    override fun onStart() {
        super.onStart()
        robot.addOnRobotReadyListener(this)
    }

    override fun onRobotReady(isReady: Boolean) {
        if (isReady) {
            try {
                val activityInfo = packageManager.getActivityInfo(componentName, PackageManager.GET_META_DATA)
                robot.onStart(activityInfo)
            } catch (e: PackageManager.NameNotFoundException) {
                throw RuntimeException(e)
            }
        }
    }
```

closed 2020-08-04 by icnocop