robotemi/sdk #109: I want Temi Assistant to support Vietnamese. Can I implement it?

lecong, 2020-08-07:
Hi Team,

I want implement for Temi Assistant support Vietnamese. How to implement it?

Thanks,

zjn0505, 2020-08-13:
In the coming release of temi SDK and software update, there are options to intercept wakeup.
Thus you will be able to do your own voice recognition, and NLP.

lecong, 2020-08-14:
have plan for this feature?

zjn0505, 2020-08-16:
The SDK support of intercepting voice flow has already been released.

Take sample app for example, as a Kiosk app, once it declares the following meta data in the manifest, it will be able to take over the voice flow, handing with its own STT and NLU engine.
```
<meta-data
            android:name="@string/metadata_override_nlu"
            android:value="true" />

<meta-data
           android:name="@string/metadata_override_stt"
           android:value="true" />
```

The SDK documentation is getting updated for this and the rest major changes of this version, and the robot software will be available in the next OTA update.

closed 2020-09-18 by lcgao

s2nmt, 2024-06-07:
Do you have a solution for this yet? Can you show me?