robotemi/sdk #111: 請問可以使用react-native開發嗎?

Fongliang, 2020-08-18:
如果要使用react-native開發的話要如何連接api

lcgao, 2020-08-20:
你好，如果reat-native有途径可以使用Android的第三方依赖库，那就是可以的。

Fongliang, 2020-08-24:
您好，
請問為什麼 adb install [option] PATH_OF_APK 我的app沒有出現在temi裡，我用一樣的方法安裝在我的手機中是成功的。
另外我的temi版本是113.23官網顯示已經更新到117我使用temi內建的軟件更新卻顯示無更新項目

lcgao, 2020-08-25:
你好，请集成 temi SDK，并在AndroidManifest.xml中的 `<application>` 元素下添加如下代码：
``` xml
<meta-data
        android:name="com.robotemi.sdk.metadata.SKILL"
        android:value="@string/app_name" />
```

zjn0505, 2020-08-25:
> 另外我的temi版本是113.23官網顯示已經更新到117我使用temi內建的軟件更新卻顯示無更新項目

请在 temi 移动 app 中联系技术支持。提供问题及序列号。

Fongliang, 2020-08-25:
> 你好，请集成 temi SDK，并在AndroidManifest.xml中的 `<application>` 元素下添加如下代码：
> 
> ```
> <meta-data
>         android:name="com.robotemi.sdk.metadata.SKILL"
>         android:value="@string/app_name" />
> ```

請問我們開發的apk一定要使用temi的sdk才能在temi中顯示嗎?
還是在AndroidManifest.xml中的 <application> 加入code後就會顯示

lcgao, 2020-08-27:
不集成 SDK 直接填入相关 `meta-data` 也可以，`android:value="@string/app_name"` 可以直接使用，用 Android Studio 创建 Android 项目时一般会默认创建这个 `String` 资源，实际值就是你的 App 名。

Fongliang, 2020-09-02:
請問安裝完成後會出現在哪個頁面，新增
<meta-data
        android:name="com.robotemi.sdk.metadata.SKILL"
        android:value="@string/app_name" />
後仍然找不到app的位置

closed 2020-09-04 by lcgao