robotemi/sdk #112: how to change Temi Screensaver with custom screen

Sandeepkrjain, 2020-08-26:
Need to have custom screensaver on Temi for corporate messaging ?

lcgao, 2020-08-27:
Hello, please consider using the [Kiosk mode](https://github.com/robotemi/sdk/wiki/Kiosk-Mode).

manhaha, 2020-08-30:
Hi @lcgao , after I turned on the Kiosk mode and select my app, Temi still display a black screen and time clock after a period of time. Is it normal?

Can i turn the time clock screensaver off in kiosk mode and let the kiosk app always on? Thanks a lot.

lcgao, 2020-08-30:
Sure, go to *Settings > General > Stand by* to turn off the screensaver.

manhaha, 2020-08-30:
Thx @lcgao , you are life saver.

May I ask one more temi kiosk app question, which is how to hide the Android navigation bar?? I set my app as kiosk app, and I tried to call the following code in MainActivity OnCreate/OnResume/OnStart, but no luck. Can you give me a little hint?

`    public void FullScreencall() {
        if(Build.VERSION.SDK_INT < 19){
            View v = this.getWindow().getDecorView();
            v.setSystemUiVisibility(View.GONE);
        } else {
            //for higher api versions.
            View decorView = getWindow().getDecorView();
            int uiOptions = View.SYSTEM_UI_FLAG_HIDE_NAVIGATION | View.SYSTEM_UI_FLAG_IMMERSIVE_STICKY;
            decorView.setSystemUiVisibility(uiOptions);
        }
    }`

lcgao, 2020-08-30:
Do you mean the navigation bar on the top of screen? If so, there are two solution to solve it:
- Swipe up in the top bar directly.
-  Use `Robot.getInstance().hideTopBar()` to hide it programmatically.

Hope this help.

manhaha, 2020-08-30:
Hi @lcgao 

No,  I mean the native android navigation bar, with back, home and rectangle button like:

[image]

My kiosk app had the top bar hidden as you suggested, but a floating half transparent android navigation bar always be there. Any way to hide it?

lcgao, 2020-08-30:
I think this is a external Navigation bar App that installed by user, so just `force-stop` this App to remove the floating navigation bar.

manhaha, 2020-08-30:
Hi @lcgao ,

Oh I see, thanks a lot! It truely is.

closed 2020-09-04 by lcgao