robotemi/sdk #115: ToggleNavigationBillboard not working

jesusonoro, 2020-08-27:
Function robot.toggleNavigationBillboard(true); doesn't disable the navigation billboard.
This doesn't works on the sample-sdk. Is there something else to do for the method to work?

lcgao, 2020-08-27:
Hello, please let me know the versions of your Launcher OS  & SDK. If you use the latest temi SDK(0.10.70) & Launcher(13582+), you need [request the Settings permission](https://github.com/robotemi/sdk/wiki/permission#requestPermissions) and set your App as the Kiosk App first. And if your Launcher is not the latest, please use the old version temi SDK (0.10.65) instead.

jesusonoro, 2020-08-27:
Hi, my Temi's Launcher OS is 13377 and I'm using the latest SDK, i will try using the old version temi sdk then. Thank you!

jesusonoro, 2020-08-28:
Using the old version temi SDK solved the problem. Thanks!

closed 2020-08-28 by jesusonoro