robotemi/sdk #117: getMapData() error

SebastianLiando, 2020-09-02:
Hi, I encountered this error when calling `Robot.getInstance().getMapData()`
>E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 116)
>E/Robot: getMapDataString() error

I followed the code exactly as the sample code provided.

The temi robot has been updated to the latest version and the map editor in temi also shows that there is a map available. Permission for the map has been granted as well. I googled about the failed binder transaction error, and my first suspect is that the map data is too big.

Could you help me with this? Thank you!

lcgao, 2020-09-03:
Yes, you are right. This method currently(0.10.70) only supports relatively small map, we will support large map in the next version. 

Sorry for the inconvenience.

SebastianLiando, 2020-09-03:
Noted, thank you!

lcgao, 2020-09-22:
Fixed this issue in version 0.10.71. Update Launcher OS to version 13716 at the same time.

closed 2020-09-22 by lcgao