robotemi/sdk #118: Firebase connection

jesusonoro, 2020-09-02:
Hello!

I'm trying to connect my Temi app with Firebase. When running, this error appears:

"Google Play services out of date.  Requires 12451000 but found 9083436"

Is it possible to update Google Play services for Temi?

Thiago-LM, 2020-09-02:
You can login on Google Play Store using a Google Account on temi. Use ADB to open the Play Store: `adb shell am start com.android.vending` then login and the Google Play Services will update itself.

jesusonoro, 2020-09-02:
> You can login on Google Play Store using a Google Account on temi. Use ADB to open the Play Store: `adb shell am start com.android.vending` then login and the Google Play Services will update itself.

Hello Thiago, I got this error while updating... 
[image]

Thiago-LM, 2020-09-02:
> > You can login on Google Play Store using a Google Account on temi. Use ADB to open the Play Store: `adb shell am start com.android.vending` then login and the Google Play Services will update itself.
> 
> Hello Thiago, I got this error while updating...

I'm not with temi right now, but I belive if you try again it will eventually work. Or maybe reboot your temi.

jesusonoro, 2020-09-02:
> I'm not with temi right now, but I belive if you try again it will eventually work. Or maybe reboot your temi.

Ok, it stopped as shown at the image above but it updated anyway... it seems, I'm testing right now

closed 2020-09-18 by lcgao