robotemi/sdk #121: 安裝SDK後app停止運作

Fongliang, 2020-09-15:
您好，我將將sdk-0.10.70.aar import後 app會停止運作如下圖
[image]
想請問可能會是發生什麼問題，是因為我這邊的temi是中國版的嗎?或是code有問題，想請問有提供使用sdk的sample code嗎?

Fongliang, 2020-09-17:
附上我用來測試SDK是否順利import的測試用code

[image]

lcgao, 2020-09-17:
你好，麻烦提供一下崩溃（crash）日志，以及temi版本号（设置>关于）。

Fongliang, 2020-09-17:
[image]
[image]

lcgao, 2020-09-17:
你好，这个temi的版本比较旧了，无法使用最新版本的SDK。目前解决方法有：

1. 联系我们的技术支持或者分销商进行版本升级。
2. 继续使用旧版本机器，同时使用旧版本SDK：`implementation 'com.robotemi:sdk-china:0.10.49'`。

推荐升级到最新版本。

Fongliang, 2020-09-17:
[image]
我後來有改用sdk-china-0.10.49.aar也是不行

lcgao, 2020-09-17:
改成0.10.49出现的崩溃的日志麻烦也提供一下

Fongliang, 2020-09-17:
[image]

lcgao, 2020-09-17:
这个.aar是自己手动编译的吧？通过 `implementation 'com.robotemi:sdk-china:0.10.49'` 集成吧。

Fongliang, 2020-09-17:
是把aar放在libs中然後再下圖dependencies區域加入implementation 'com.robotemi:sdk-china:0.10.49'嗎?
[image]

lcgao, 2020-09-17:
是的

Fongliang, 2020-09-17:
好的感謝您的幫忙我再試試看

Fongliang, 2020-09-17:
> 你好，这个temi的版本比较旧了，无法使用最新版本的SDK。目前解决方法有：
> 
> 1. 联系我们的技术支持或者分销商进行版本升级。
> 2. 继续使用旧版本机器，同时使用旧版本SDK：`implementation 'com.robotemi:sdk-china:0.10.49'`。
> 
> 推荐升级到最新版本。

請問如果要進行版本更新一定要經過分销商嗎?

lcgao, 2020-09-18:
或者通过这里：https://www.robotemi.com/helpdesk/  来联系我们的技术支持。

Fongliang, 2020-09-18:
> 或者通过这里：https://www.robotemi.com/helpdesk/ 来联系我们的技术支持。

如果經由這個方式的話可以不將機器人送回，自行將temi進行版本更新嗎?
是透過下圖的地方進行聯繫嗎?請問有中文的技術支持嗎?
感謝您的幫忙!
[image]

lcgao, 2020-09-18:
不需要寄回机器人，会有中文技术支持安排推送更新的。

Fongliang, 2020-09-18:
> 不需要寄回机器人，会有中文技术支持安排推送更新的。

適用下圖的地方申請支持嗎?
[image]

lcgao, 2020-09-18:
https://robotemi.zendesk.com
在这里注册账号提交工单吧

Fongliang, 2020-09-18:
好的感謝您!

closed 2020-09-18 by Fongliang