robotemi/sdk #126: skidJoy API does not control speed smoothly

Jeff-Zhu, 2020-09-21:
I am trying to use the skidJoy to Temi turning radius, but regardless what other values I used, it always seems to be treated as 0, or 1. 
Such as skidJoy(0.5, 0) behaves as same as skidJoy(1,0). 
And skidJoy(0.5, 1) is same as skidJoy(1,1).
And skidJoy(0.25, 1) is actually only turning.

BTW, I do have to repeat the skdJoy every 500ms, otherwise Temi won't move.

Is it possible to change turning radius on the move continuously?

lcgao, 2020-10-19:
Please consider using [`Robot.getInstance().turnBy(int degrees)`](https://github.com/robotemi/sdk/wiki/Movement#turnBy).

zjn0505, 2020-10-19:
> BTW, I do have to repeat the skdJoy every 500ms, otherwise Temi won't move.

This is expected.

root221, 2020-12-02:
What is the difference between skidjoy(0.5,0) and skidjoy(1,0)

lcgao, 2021-07-22:
Using `skidJoy(0.5, 0)` will move forward temi with 0.5 * max speed. `skidJoy(1, 0)` will move with the max speed.

closed 2021-07-22 by lcgao