robotemi/sdk #130: onBatteryStatusChanged 以及地圖保存 問題

Fongliang, 2020-10-05:
你好，
Q1:想請問當電池電量發生變化時onBatteryStatusChanged 這個l這個监听器會接收到嗎?因為如下圖所示當電量發生變化(從92->91時沒有回傳任何log)，Q2:想要請問temi可以存多個樓層的地圖嗎?(例如一次存一樓和二樓的地圖然後可以藉由電梯在多樓層中移動)
[image]
[image]

zjn0505, 2020-10-05:
目前只有在同一平面、連續的地圖內才有最佳的導航和定位效果。
我們在最近的一次版本更新中提供了 temi center 的服務，可以隨時將地圖備份至雲端，及將雲端地圖導入到機器本體。
這樣可以解決一部分需要在非連續地圖之間切換的場景。

更多場景的支持會在之後的迭代中實現。

Fongliang, 2020-10-05:
> 目前只有在同一平面、連續的地圖內才有最佳的導航和定位效果。
> 我們在最近的一次版本更新中提供了 temi center 的服務，可以隨時將地圖備份至雲端，及將雲端地圖導入到機器本體。
> 這樣可以解決一部分需要在非連續地圖之間切換的場景。
> 
> 更多場景的支持會在之後的迭代中實現。

請問timer center的更新已經推送了嗎?

Fongliang, 2020-10-05:
另外請問我想要使用setNavigationSafety去更改SafetyLevel但是沒有作用 如下圖
[image]

----------------------log------------------------------------------

[image]

zjn0505, 2020-10-05:
如果是上一條 ticket #121 中的機器人，現在應該已經推送了最新版本，並支持 temi center 的使用了。
請訪問 https://center.robotemi.cn 管理您的機器人。
`setNevigationSafety` 的方法請對應文檔 [#setNavigationSafety](https://github.com/robotemi/sdk/wiki/Utils#setNavigationSafety)
檢查該應用是否為當前選中的 kiosk 商戶模式應用，以及是否賦予了 Settings 設置的權限。

Fongliang, 2020-10-06:
[image]
[image]
kiosk 商戶模式應用，以及Settings 設置的權限都已經開啟但仍然無法使用，使用toast看checkSelfPermission以及isSelectedKioskApp都回傳沒有權限

lcgao, 2020-10-12:
你好，因为这涉及到跨进程调用，实际生效需要一定的时间，所以并不能立刻在下一行代码中获取到修改后的结果。请在设置安全等级后在设置>导航>避障灵敏度中查看是否修改成功。

Fongliang, 2020-10-12:
> 目前只有在同一平面、連續的地圖內才有最佳的導航和定位效果。
> 我們在最近的一次版本更新中提供了 temi center 的服務，可以隨時將地圖備份至雲端，及將雲端地圖導入到機器本體。
> 這樣可以解決一部分需要在非連續地圖之間切換的場景。
> 
> 更多場景的支持會在之後的迭代中實現。

請問導入地圖是一定要透過網站導入嗎?可以讓機器人在更換樓層時透過api自己導入新地圖嗎?

Fongliang, 2020-10-12:
> 你好，因为这涉及到跨进程调用，实际生效需要一定的时间，所以并不能立刻在下一行代码中获取到修改后的结果。请在设置安全等级后在设置>导航>避障灵敏度中查看是否修改成功。
好的感謝

zjn0505, 2020-10-14:
> > 目前只有在同一平面、連續的地圖內才有最佳的導航和定位效果。
> > 我們在最近的一次版本更新中提供了 temi center 的服務，可以隨時將地圖備份至雲端，及將雲端地圖導入到機器本體。
> > 這樣可以解決一部分需要在非連續地圖之間切換的場景。
> > 更多場景的支持會在之後的迭代中實現。
> 
> 請問導入地圖是一定要透過網站導入嗎?可以讓機器人在更換樓層時透過api自己導入新地圖嗎?

目前是这样，只能透过网站导入。我们会逐步将地圖操作开放给第三方应用。

Fongliang, 2020-10-15:
好的感謝解答!

closed 2020-10-15 by Fongliang