robotemi/sdk #134: How we can Run Temi SDK on Mobile?

NumanGillani-jackys, 2020-10-21:
We've one Application for Mobiles, and the same we'll use for Temi. Now I integrated the Temi SDK in my application, and it's working fine on Temi. 

The problem is now when I open the application in any other device, it crashed. Only working on Temi, not any other device.

lcgao, 2020-10-23:
All features related temi SDK should run with real temi robot, otherwise your application will be forced stop by temi SDK, anyway you can refer to #90 to solve this potential problem.

NumanGillani-jackys, 2020-11-05:
That problem is fixed. Right now I am facing another issue.

Whenever I install the application from Android Studio, it is installed but don't open automatically. Each time I've to go and click on App Icon to start the Application. Usually whenever we install the apk, it launches on device but for this case it is not launching.

Any idea about it?

DanLechuga00, 2021-08-07:
Then you need to empty the cache memory of the application, Android Studio sends a message in the cat log of the application that has been installed, but it cannot be started

DanLechuga00, 2021-08-07:
But if I don't have the robot to test my code, I can do code tests, fake listeners, or just test the functionality, without involving the SDK