robotemi/sdk #138: Hide Temi Top Bar from Application

mmorisco, 2020-11-10:
Hello, I am trying to hide the Temi Top Bar of Time/Battery/Settings etc from my Application programmatically. I tried to use what you suggested in issue #89, but it doesn't works. 

This is androidManifest:
 <activity
            android:name="com.predict.speech.aphel.util.WebActivity"
            android:label="@string/app_name"
            android:theme="@style/AppTheme.NoActionBar" >

            <meta-data
                android:name="com.robotemi.sdk.metadata.UI_MODE"
                android:value="4" />
        </activity>

And this is my "RobotOnReady" :

    @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {
                final ActivityInfo activityInfo = getPackageManager().getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }

lcgao, 2020-11-11:
Hello, did you add the OnRobotReadyListener to your context like [here](https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L174)?

mmorisco, 2020-11-11:
Yes, sure. I put a log in the "onRobotReady" and I see it in the logcat. But the only effect I see is the hiding of the topbar, but not completely hide (a "dash" remains at the top of the screen and if I click the dash I see the topbar)

lcgao, 2020-11-11:
Check if any where use the `robot.hideTopBar()` method and that will make the "dash" visible.

mmorisco, 2020-11-11:
Hi! no use of robot.hideTopBar() or robot.showTopBar() in all of my project. 
This is my AndroidManifest.xml... Where I have to put the meta element according to you?
Please note that all Activities extends the base class "com.predict.speech.aphel.util.WebActivity"

<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.predict.speech.aphel">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
    <uses-permission android:name="android.permission.CHANGE_NETWORK_STATE" />
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW" />
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>

    <uses-permission android:name="com.ubtrobot.permission.ROBOT" />

    <application
        android:name="com.predict.speech.aphel.App"
        android:allowBackup="true"
        android:hardwareAccelerated="false"
        android:icon="@drawable/iconpredict"
        android:label="@string/app_name"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">
        <activity
            android:name="com.predict.speech.aphel.view.LoginActivity"
            android:label="@string/app_name"
            android:theme="@style/AppTheme.NoActionBar">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
        <activity
            android:name="com.predict.speech.aphel.speech.SurveyActivity"
            android:label="@string/title_activity_speech"
            android:windowSoftInputMode="adjustPan"
            android:theme="@style/AppTheme.NoActionBar" />
        <activity
            android:name="com.predict.speech.aphel.speech.SpeechActivity"
            android:label="@string/app_name"
            android:windowSoftInputMode="adjustPan"
            android:theme="@style/AppTheme.NoActionBar"/>

        <activity
            android:name="com.predict.speech.aphel.consenso.ConsensoActivity"
            android:label="@string/app_name"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustPan"
            android:theme="@style/AppTheme.NoActionBar"/>
        <receiver
            android:name="com.predict.speech.aphel.ActivityforStartup"
            android:enabled="true"
            android:exported="true"
            android:permission="android.permission.RECEIVE_BOOT_COMPLETED">
            <intent-filter>
                <action android:name="android.intent.action.BOOT_COMPLETED" />
                <action android:name="android.intent.action.QUICKBOOT_POWERON" />
[... 53 lines cut ...]
        <meta-data
            android:name="com.robotemi.sdk.metadata.ACTIONS"
            android:value="home.welcome,
                        home.dance,
                        home.sleep" />

        

        <meta-data
            android:name="com.robotemi.sdk.metadata.CONTEXTS"
            android:value="
                radio,
                media" />

        <meta-data
            android:name="@string/metadata_kiosk"
            android:value="TRUE" />

        <meta-data
            android:name="@string/metadata_open_without_internet"
            android:value="TRUE" />

    </application>

</manifest>

lcgao, 2020-11-11:
``` java
<activity
        android:name="com.predict.speech.aphel.view.LoginActivity"
        android:label="@string/app_name"
        android:theme="@style/AppTheme.NoActionBar">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />

            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>

        <meta-data
            android:name="com.robotemi.sdk.metadata.UI_MODE"
            android:value="4" />
</activity>
```
If it still doesn't work, please consider adding the meta-data to all activities.

mmorisco, 2020-11-11:
Great! It works!

mmorisco, 2020-11-11:
Thank you very much for the support

closed 2020-11-11 by mmorisco

quil-cityu, 2020-11-19:
Hi @lcgao,
I followed your guideline and cannot hide the top bar. The small "dash" is visible on top.

My application is opened in Kiosk Mode. Because it is a flutter application. I only have one Activity XML in the manifest file. This is my manifest file

```xml
<manifest xmlns:android="http://schemas.android.com/apk/res/android" package="hk.edu.cityu.temi">
  <!-- io.flutter.app.FlutterApplication is an android.app.Application that
         calls FlutterMain.startInitialization(this); in its onCreate method.
         In most cases you can leave this as-is, but you if you want to provide
         additional functionality it is fine to subclass or reimplement
         FlutterApplication and put your custom class here. -->
  <application android:name="io.flutter.app.FlutterApplication" android:label="mobile_flutter_temi" android:icon="@mipmap/ic_launcher">
    <activity android:name=".MainActivity" android:launchMode="singleTop" android:theme="@style/LaunchTheme" android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode" android:hardwareAccelerated="true" android:windowSoftInputMode="adjustResize">
      <!-- Specifies an Android theme to apply to this Activity as soon as
                 the Android process has started. This theme is visible to the user
                 while the Flutter UI initializes. After that, this theme continues
                 to determine the Window background behind the Flutter UI. -->
      <meta-data android:name="io.flutter.embedding.android.NormalTheme" android:resource="@style/NormalTheme" />
      <!-- Displays an Android View that continues showing the launch screen
                 Drawable until Flutter paints its first frame, then this splash
                 screen fades out. A splash screen is useful to avoid any visual
                 gap between the end of Android's launch screen and the painting of
                 Flutter's first frame. -->
      <meta-data android:name="io.flutter.embedding.android.SplashScreenDrawable" android:resource="@drawable/launch_background" />
      <meta-data android:name="com.robotemi.sdk.metadata.UI_MODE" android:value="4" />
      <intent-filter>
        <action android:name="android.intent.action.MAIN" />
        <category android:name="android.intent.category.LAUNCHER" />
      </intent-filter>
    </activity>
    <!-- Don't delete the meta-data below.
             This is used by the Flutter tool to generate GeneratedPluginRegistrant.java -->
    <meta-data android:name="flutterEmbedding" android:value="2" />

    <!-- Adding this meta-data is required in order to display this app in temi's launcher app -->
    <meta-data android:name="com.robotemi.sdk.metadata.SKILL" android:value="CityU Temi" />

    <!--
    This meta-data is telling the launcher which nlp responses should
    invoke the NlpResponse listener's onNlpResponse method sending it these intents
    -->
    <meta-data android:name="com.robotemi.sdk.metadata.ACTIONS" android:value="home.welcome,home.sleep" />

    <meta-data android:name="com.robotemi.sdk.metadata.KIOSK" android:value="true" />
  </application>

  <uses-permission android:name="android.permission.INTERNET" />
  <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
</manifest>
```

Thank you very much.