robotemi/sdk #140: OnTelepresenceStatusChangedListener 無法使用

Fongliang, 2020-11-16:
您好，
我在新增OnTelepresenceStatusChangedListener會出現下列錯誤，，顯示 interface expected here
[image]

lcgao, 2020-11-16:
您好，就如错误提示所说，`implements` 后面接的应该是 `interface` (接口)，但是 `OnTelepresenceStatusChangedListener` 是个 `class` (类)，不能接在 `implements` 后面。

Fongliang, 2020-11-16:
請問如果不是interface我要怎麼宣告OnTelepresenceStatusChangedListener?以及如何讓OnTelepresenceStatusChangedListener監聽
TelepresenceStatus，直接Robot.getInstance().addOnTelepresenceStatusChangedListener();嗎?

lcgao, 2020-11-17:
如果满足需求的话，建议使用 [`interface OnTelepresenceEventChangedListener {}`](https://github.com/robotemi/sdk/wiki/Users-&-Telepresence_chn#addOnTelepresenceEventChangedListener)，这样就可以和其他监听器接口一样使用了。

Fongliang, 2020-11-17:
好的感謝您!

closed 2020-11-17 by Fongliang