robotemi/sdk #143: Screen Recording

SebastianLiando, 2020-11-25:
Hi, is there a way to record temi's screen for better documentation purpose?

Android Studio's screen record feature stops immediately after starting. Also tried with scrcpy, but an exception is thrown. Mobizen app can not even run in temi.

lcgao, 2020-11-30:
Hi, we will release new version for Android OS firmware to solving this kind of issue in the future.

zjn0505, 2020-12-10:
> Hi, is there a way to record temi's screen for better documentation purpose?
> 
> Android Studio's screen record feature stops immediately after starting. Also tried with scrcpy, but an exception is thrown. Mobizen app can not even run in temi.

Can you check what will happen if using
> scrcpy -m 1024

SebastianLiando, 2020-12-10:
> > Hi, is there a way to record temi's screen for better documentation purpose?
> > Android Studio's screen record feature stops immediately after starting. Also tried with scrcpy, but an exception is thrown. Mobizen app can not even run in temi.
> 
> Can you check what will happen if using
> 
> > scrcpy -m 1024

Hi, thanks for the suggestion.

Yup, it is now able to cast temi's screen. However, it seems like the tool can not manage the orientation properly. As attached, the screen is cast vertically, but the controls work horizontally (swiping left and right works although the screen is vertical). 

[image]

zjn0505, 2020-12-10:
Thanks for the info. temi's next Android firmware OTA will fix this rotation issue.
We will rollout this OTA gradually.

Please contact temi customer support and provide the robot serial number.
It can be found on the robot under `Settings -> About -> Serial Number`.

zjn0505, 2020-12-24:
OTA firmware ver. 20201216 is rolling out to fix this issue.

Thiago-LM, 2021-05-27:
I can confirm that `scrcpy -m 1024` works now on latest firmware.

[image]

closed 2021-07-22 by lcgao