robotemi/sdk #145: Overriding the onRobotReady method

LeoFau, 2020-11-26:
Hello I am pretty new to programming in general and trying to create an app on the temi robot in course of writing an essay for college.

I tried to follow the instructions on starting as explained in the wiki. When I paste in the paragraph on overriding the onRobotReady method, Android Studio does not recognize the "robot"-parameter? What do I need to do, to change this?
[image]

Thank you in advance!

Leo

lcgao, 2020-11-27:
Hello, please replace `robot.onStart(activityInfo)` to `Robot.getInstance().onStart(activityInfo)`.

LeoFau, 2020-11-27:
Thank you, that worked!

closed 2020-11-27 by LeoFau