robotemi/sdk #146: ASR Listener, Tts Request not executing

Robotrator, 2020-11-30:
I am using ASR Listener for custom Commands. It is working absolutely fine, askQuestion is going very well. All commands are being executed perfectly except robot.speak.

I checked the Logs, focus goes from each statement, but not reflecting on Temi Screen. NLU is enabled from Manifest, Kiosk Mode is also enabled. Below is the statement, which is not working and generating zero response. Even I made 2nd parameter as True so I can see the text on screen but text doesn't appear.

robot.speak(TtsRequest.create("Okay, please enjoy.", false));

Could you please guide me the steps to make it working.

KaoruKZ, 2020-12-01:
Sorry for bad english and I hope I understand your question correctly. Are you trying to use robot.speak() after robot.askQuestion()? If yes, you have to add robot.finishConversation() before you can use robot.speak().

ahackit, 2020-12-15:
Adding on to what @KaoruKZ mentioned, I'm attempting React Native bindings on top of Temi SDK and noticed that for delayed Speak events I had to start off my listener with a `finishConversation()` by default or Temi would attempt its default ASRResult.

`
@Override
  public void onAsrResult(String asrResult) {

      Robot.getInstance().finishConversation();
      
      // React Native Event Fire
      WritableMap params = Arguments.createMap();
      params.putString("asrResult", asrResult);
      sendEvent(context, "onAsrResult", params);
  }
`

closed 2021-10-19 by lcgao