robotemi/sdk #147: Get More Useful Map Data from API

DoubleWB, 2020-11-30:
Hello,

I am trying to accomplish some visual representation about the locations that are saved on the Temi in terms of their absolute positions on the map.

It appears to me that the available related information you can get from the Temi is:
- The current location of the Temi.
- The list of location names.
- The distance to a particular location.
- The bitmap image representing the map and all the locations graphically, but no information about connecting locations on the map to the location name list.

With this set of functionality, it is possible to get a set of the (x,y, theta) coordinates of each of the locations, but only by moving the Temi and collecting information at various locations. Furthermore, even if one has this list, there's no way to map labels onto the map image without using some computer vision processing on the bitmap.

If I am mistaken about how to get a list of the coordinates of the locations and then draw them onto the map, I would very much appreciate some guidance on how to do this! Otherwise, I would argue for the addition of this functionality to the API, because the image of the map without any ability to know how to map real life locations onto it is not very useful.

Thank you in advance!

lcgao, 2021-01-05:
Hello, we are working on this and it will be released in next version, thanks.

lcgao, 2021-07-22:
Hello, developers can get more information of map data from [version 0.10.74](https://github.com/robotemi/sdk/releases/tag/v0.10.74).

closed 2021-07-22 by lcgao