robotemi/sdk #152: Help needed with Kiosk Mode

hybrid87sg, 2020-12-23:
I need help on understanding how to set my custom app as kiosk mode

relatively new to coding and been using Kotlin as my language. Trying to figure out how to convert the Java source to Kotlin language

lcgao, 2020-12-24:
Please add following code to the **AndroidManifest.xml** as a child of `<application>` element,

``` xml
<meta-data
    android:name="com.robotemi.sdk.metadata.SKILL"
    android:value="@string/app_name" />

<meta-data
    android:name="com.robotemi.sdk.metadata.KIOSK"
    android:value="true" />
```

And then go to *Settings > App > Kiosk*, turn on the switch and select your App.

hybrid87sg, 2020-12-26:
another question is how do I actually install my custom app as a Temi app?

if I were to wireless adb,I cannot access my app again unless I go through android studio to do it again

closed 2020-12-29 by hybrid87sg