robotemi/sdk #160: How to trigger another function when Temi arrive at a location

rafonsosilva, 2021-01-11:
If I want Temi to go to several locations at once, what would be the best approach?
For example, I want to create a tour mode for Temi. When the tour mode is called, Temi would go to location A and after arriving at location A, say "this is location A" and then go automatically to location B, and at the arrival say "this is location B"; at the end of the tour, Temi would go automatically to its base.

lcgao, 2021-01-11:
Hi, please see this interface: https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener.

Listen to this listener and your App will receive the 'complete' event of 'go to A', trigger another 'go to B' once receive that event.

AK51, 2021-05-31:
Hi, I an new to Temi. I can make the Temi to go to certain location. But how to set the Temi to say something after it is arrived?
Not really sure how to use this listerner...
Is there any sample code for reference? Thx

lcgao, 2021-05-31:
Yes, we have the sample code in this project, please checkout following link.
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L67
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L120
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L149
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L726-L737

AK51, 2021-06-03:
Hi, I am still working on it... (I am not a good android java developer. >_<)
mListener.equals("complete") is wrong
May I know what is a proper way to check the completeness? Thx

import com.robotemi.sdk.listeners.OnGoToLocationStatusChangedListener;
...

public class MainActivity extends AppCompatActivity {
    private OnGoToLocationStatusChangedListener mListener;
...
            if(mListener.equals("complete")){...}

closed 2021-07-22 by lcgao