robotemi/sdk #163: Get Map Data and Robot Position

LeoFau, 2021-01-14:
Hello,

I am still very new to programming android apps.
I am trying to get the map data as well as the robot's position. I have read the sample code for getting the map data and tried to implement it into my app, but android studio seamed pretty unhappy with the input. Especially the imports concerning kotlin aspects seem to be a problem. Do I have to implement or import something else before I can use the method? 

I would also like to get the robot's position. I read that I would have to use  OnCurrentPositionChangedListener, but I am having trouble understanding how this is done. Can you help me?

Thank you very much.

Leo

lcgao, 2021-01-19:
Hello,

Your App needs the MAP permission before using the method to get the map data, check the documentation to learn how to grant permission to your App: https://github.com/robotemi/sdk/wiki/permission.

You mean you can't listen to the changes of current positon? If so please refer to the sample code:
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L108
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L200
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L974-L977

Hope this help.

LeoFau, 2021-01-22:
Okay thank you. The sample code did help me a lot. I am slightly stuck on an error I get on multiple methods after copying the sample code as my starting point (see attached picture). How can I resolve this issue? It is the same issue for multiple other methods too.

Thank you again. I know my questions are probably very basic.
[image]

lcgao, 2021-07-22:
As the IDE mentioned, there are two function using the same function name and the same parameter `onDetectionStateChanged(in state)`, so just keep only one of them.

closed 2021-07-22 by lcgao