robotemi/sdk #171: Temi top bar keep turning back

ThiagoHBA, 2021-01-29:
Hello, i need to remove the native topbar from Temi. i already have seen the issue #89, and work fine for me. But the problem is: When i leave my aplicattion and go back to the Temi system, or active the standby mode, she keeps comming back. Looking for solutions in the program I came across with some functions in the file `FlutterActivity.java` I noticed that when exiting the flutter application it activates the onPause function and when returning it activates the onResume and onPostResume function. After realizing this I tried to use the RefreshTemiUi function that was mentioned in issue #89, but I was not successful.

my refreshTemiUi function:
```
    private void refreshTemiUi() {
        try {
            Log.d(TAG, "Refresh temi UI");
            ActivityInfo activityInfo = getPackageManager()
                    .getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
           robot.onStart(activityInfo);
        } catch (PackageManager.NameNotFoundException e) {
            e.printStackTrace();
        }
    }
```
my onResume and onPostResume function

```
    @Override
    protected void onResume() {
        super.onResume();
        refreshTemiUi();
        Log.d(TAG, "on resume");
    }

    @Override
    public void onPostResume() {
        super.onPostResume();
        refreshTemiUi();
        Log.d(TAG, "on post resume");
    }
```
i put the logs just to see when the function was called, 
that way I know what function is called after going back to the application.

My `AndroidManifest.xml ` is the same as the code in the question mentioned above, would anyone have any idea how to fix the problem?

lcgao, 2021-02-03:
Hello, if possible please show your AndroidManifest.xml here.

ThiagoHBA, 2021-02-03:
My AndroidManifest.xml file : 

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.souenergy_salespromoter">

    <!-- Permissões necessario para WiFi -->
    <uses-permission android:name="android.permission.CHANGE_WIFI_STATE"/>
    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE"/>
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-feature android:name="android.hardware.wifi" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

    <uses-permission android:name="android.permission.KILL_BACKGROUND_PROCESSES" />

    <application
        android:label="souenergy_salespromoter"
        android:icon="@mipmap/ic_launcher">

        <!--
        Adding this meta-data is required in order to display this app in temi's launcher app.
        -->
        <meta-data
            android:name="com.robotemi.sdk.metadata.SKILL"
            android:value="Promotor de Vendas" />
        <!--
        Adding this meta-data is required in order to display this app in temi's kiosk mode.
        -->
        <meta-data
            android:name="com.robotemi.sdk.metadata.KIOSK"
            android:value="TRUE" />
        <!--
        Adding this meta-data is required in order to use this app in temi without internet.
        -->
        <meta-data
            android:name="com.robotemi.sdk.metadata.OPEN_WITHOUT_INTERNET"
            android:value="TRUE" />

        <meta-data
            android:name="@string/metadata_permissions"
            android:value="com.robotemi.permission.map,
                            com.robotemi.permission.face_recognition,
                            com.robotemi.permission.settings,
                            com.robotemi.permission.sequence" />

        <activity
            android:name=".MainActivity"
            android:launchMode="singleTask"
            android:theme="@style/LaunchTheme"
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustResize">
            <!-- Specifies an Android theme to apply to this Activity as soon as
                 the Android process has started. This theme is visible to the user
                 while the Flutter UI initializes. After that, this theme continues
                 to determine the Window background behind the Flutter UI. -->
            <meta-data
                android:name="io.flutter.embedding.android.NormalTheme"
                android:resource="@style/NormalTheme"
                />
            <!-- Displays an Android View that continues showing the launch screen
                 Drawable until Flutter paints its first frame, then this splash
                 screen fades out. A splash screen is useful to avoid any visual
                 gap between the end of Android's launch screen and the painting of
                 Flutter's first frame. -->
            <meta-data
                android:name="io.flutter.embedding.android.SplashScreenDrawable"
                android:resource="@drawable/launch_background"
                />
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>

            <meta-data
                android:name="com.robotemi.sdk.metadata.UI_MODE"
                android:value="4" />
            <!--
            <meta-data
                android:name="@string/metadata_permissions"
                android:value="com.robotemi.permission.map" />
                -->

        </activity>

        <!-- Don't delete the meta-data below.
             This is used by the Flutter tool to generate GeneratedPluginRegistrant.java -->
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
</manifest>

```

lcgao, 2021-02-07:
I didn't find the declaration of `FlutterActivity.java` in AndroidManifest.xml, did you do anything related the top bar in `FlutterActivity.java`?

ThiagoHBA, 2021-02-07:
The `MainActivity.java` file extends to the `FlutterActivity.java` file, where we can find the functions of `onResume, onPostResume, onPause, onCreate`, etc. 

my MainActiviy.java: 

```

public class MainActivity extends FlutterActivity implements OnRobotReadyListener {
    FlutterEngine engine;
    BinaryMessenger binaryMessenger;
    TemiRobotHandler temiRobotHandler;
    Context maCtx;
    private Robot robot = Robot.getInstance();

    private void refreshTemiUi() {
        try {
            ActivityInfo activityInfo = getPackageManager()
                    .getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
           robot.onStart(activityInfo);
        } catch (PackageManager.NameNotFoundException e) {
            e.printStackTrace();
        }
    }

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        maCtx = getApplicationContext();
        binaryMessenger = engine.getDartExecutor().getBinaryMessenger();
        this.temiRobotHandler = new TemiRobotHandler(binaryMessenger, maCtx);
        robot.addOnRobotReadyListener(this);
        refreshTemiUi();
        robot.addOnBatteryStatusChangedListener(this.temiRobotHandler);
        robot.addOnDetectionStateChangedListener(this.temiRobotHandler);
        robot.addOnGoToLocationStatusChangedListener(this.temiRobotHandler);
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        robot.removeOnRobotReadyListener(this);
        robot.removeOnBatteryStatusChangedListener(this.temiRobotHandler);
        robot.removeDetectionStateChangedListener(this.temiRobotHandler);
        robot.removeOnGoToLocationStatusChangedListener(this.temiRobotHandler);
    }

    @Override
    public void configureFlutterEngine(@NonNull FlutterEngine flutterEngine) {
        GeneratedPluginRegistrant.registerWith(flutterEngine);
        this.engine = flutterEngine;
    }

    @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            ActivityInfo activityInfo = null;
            try {
                activityInfo = getPackageManager()
                        .getActivityInfo(getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
            } catch (PackageManager.NameNotFoundException e) {
                e.printStackTrace();
                //Log.d("onRobotReady", " - error = " + e);
            }
            //assert activityInfo != null;
            //robot.onStart(activityInfo);
        }
    }
}
```

Class FlutterActivity : https://api.flutter.dev/javadoc/io/flutter/app/FlutterActivity.html

ThiagoHBA, 2021-02-10:
It would also work if it were possible to permanently remove the native topBar from the operating system of Temi.

lcgao, 2021-02-10:
Your code looks okay but it's weird that doesn't work. The top bar is necessary for temi Launcher OS, so the top bar will be displayed when user back to Launcher.

ThiagoHBA, 2021-02-10:
My idea would be to use a command to hide the top bar just like what happens when we start the application. It's okay it appeared when we left the application.

lcgao, 2021-02-10:
Yeah it should be like that, so you just need to check the code of `MainActivity`. Try to just invoke the `refreshTemiUi()` in `onResume()`.

ThiagoHBA, 2021-02-10:
Can you give me the code of how you're imagining it would work? That of using `RefreshTemiUi()` in `onResume()` I'm already doing, is there a possibility that I have to add something else in `androidManifest.xml` for him to recognize this command in` onResume()`? Or even use some native android function.

lcgao, 2021-02-22:
You can add a button and add OnClickListener for it to test if removing top bar by `refreshTemiUi()` is useful.

closed 2021-07-22 by lcgao