robotemi/sdk #175: Receiving Push Notifications

lulumiddel, 2021-02-23:
Hello, 
I was wondering if there is any way to forward push notifications generated by regular android apps to the temi? For example, when using the messenger App "Element", the user is currently not shown if a new message arrives. 
I'm grateful for any help I can get!

lcgao, 2021-03-04:
temi disabled the native notification component by default, so please consider using other way to show the new message.

lulumiddel, 2021-03-12:
Thank you for your answer! Can you think of any other way to display the messages other than having Temi-native apps?

lulumiddel, 2021-04-01:
> 
> 
> temi disabled the native notification component by default, so please consider using other way to show the new message.

Are there any plans to include the notification center in the temi API? Or is there already a reference that I wasn't able to find?