robotemi/sdk #177: App Icon disappears from top bar when opening settings

LeoFauSR, 2021-03-11:
Hello,

I have seen issue #136 and I have integrated the code you suggested. When I close my app, the icon is still there in the top bar. But if I open the robot's settings after closing my app, the icon gets replaced by the settings icon and I can no longer access my app without running it again from android studio via the adb.

Is there a way to permanently secure the icon to the top bar or the screen of temi? 
Like I said, I already included the metadata mentioned in issue #136 in my manifest.

Thank you in advance.

Leo

lcgao, 2021-03-12:
Hello, you can access your App in temi's apps(Contacts, Locations, Settings...).

LeoFauSR, 2021-03-12:
The App is not being displayed in temi's apps. How can I change that?

lcgao, 2021-04-30:
Sorry for missing the latest message. Please add following code as the child tag of `<application>` in AndroidManifest.xml:

``` xml
<meta-data
        android:name="com.robotemi.sdk.metadata.SKILL"
        android:value="@string/app_name" />
```

LeoFauSR, 2021-04-30:
Thank you, I added the code in the wrong tag.

closed 2021-04-30 by LeoFauSR