robotemi/sdk #182: Is there a way to import maps into temi with MapDataModel?

ncostaka, 2021-03-17:
You can get the map using getMapdata().
Conversely, is there a way to import maps into temi with MapDataModel?

lcgao, 2021-03-18:
No, but developers can [import map by map ID](https://github.com/robotemi/sdk/wiki/Locations#loadMap) that from [getMapList()](https://github.com/robotemi/sdk/wiki/Locations#getMapList).

nabe899, 2021-03-23:
I have a question for the answer to this question.

I'm thinking of using multiple Temi.
I want multiple Temi to have the same map, so I think it would be good if I could distribute a map registered with one Temi to another. I understand that I can't do it now, but do you have any plans to be able to register maps taken from Temi with other Temi in the future?

lcgao, 2021-03-25:
If the temis are activated by one user, so these temis can share the backed up maps of that user.

nabe899, 2021-03-29:
Thank you for your answer.
I understood how to say how to go through the Temi center.

I hope I can do it directly with the SDK, not via the Temi Center. Can the SDK of the future do it?

zjn0505, 2021-03-31:
> I hope I can do it directly with the SDK, not via the Temi Center. Can the SDK of the future do it?

In the SDK, [getMapList()](https://github.com/robotemi/sdk/wiki/Locations#getMapList) will provide all the maps saved on temi center.
With map id from the returned list, [loadMap()](https://github.com/robotemi/sdk/wiki/Locations#loadMap) will import the target map.

> but do you have any plans to be able to register maps taken from Temi with other Temi in the future?

Combining these two methods, a map being imported can be the one uploaded from another robot under your account.

Please be aware that as each robot is calibrated individually, it is not recommended at this moment to use map from another robot. You will see this notice on temi center when sending map of another robot. The robot may encounter navigation and localization issue with map data created from another robot.

nabe899, 2021-03-31:
Thanks for your polite answer.

> Please be aware that as each robot is calibrated individually, it is not recommended at this moment to use map from another robot. 

After uploading maps from other Temi, I thought that re-registering home_base would work fine, but is my idea wrong?

zjn0505, 2021-04-01:
Yes, it is an important step to let the robot know its initial place in a newly loaded map.

In the temi [123 version](https://www.robotemi.com/software-updates/), load a new map from SDK requires the robot to be placed on home base. and in the coming 124 version, it will be more flexible, load map from anywhere plus robot can do auto re-positioning.

Using a map from other robots is letting a robot to see the world with memory of another. As the map is built from combined data of sensors and lidar calibrated specifically for each robot. So in the physical world it is not likely that every robot will see the world the same. There are chances the robot won't navigate as well as using its own map, while we are not restricting this function neither from temi center nor in SDK as we want to explore its feasibility.

nabe899, 2021-04-04:
Thank you for answering many times.
I was able to understand the current situation well.
I would also like to explore the feasibility.

VantieghemMarijn, 2021-07-13:
Hello i am develloping a mqtt app for all functionalities for temi but when i call the getMapList() in java it returns 0 maps all the time while he is mappend and i made a backup.

Anyone know why it does this or how to get the current map?

lcgao, 2021-07-22:
Please make sure that call `getMapList()` after the SDK is ready, listen to the `OnRobotReadyListener` listener to know if the SDK is ready.

closed 2021-07-22 by lcgao

VantieghemMarijn, 2021-07-22:
Okay thank you maybe i did not call the getmaplist function in the right postition.
And once again thank you for the quick response.

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

Closed #182<https://github.com/robotemi/sdk/issues/182>.