robotemi/sdk #183: How to use the onTelepresenceStatusChangedListener class?

Yusk1450, 2021-03-18:
I want to use the onTelepresenceStatusChangedListener.
However callEventModel.sessionId is empty at onTelepresenceEventChanged method.

The sessionId's Description in Github's documentation has been strikethrough.
https://github.com/robotemi/sdk/wiki/Users-&-Telepresence#onTelepresenceStatusChangedListener

Why dose it?
How to use the onTelepresenceStatusChangedListener class?

lcgao, 2021-03-19:
Please use [OnTelepresenceEventChangedListener](https://github.com/robotemi/sdk/wiki/Users-&-Telepresence#onTelepresenceEventChangedListener) interface instead.

Yusk1450, 2021-03-23:
Thanks!
Can I use OnTelepresenceEventChangedListener to get the event that the call was DECLINED ?

lcgao, 2021-07-22:
Currently can't.

closed 2021-07-22 by lcgao

luozhexin, 2022-08-08:
Can I use onTelepresenceStatusChangedListener class to monitor the call was DECLINED ?