robotemi/sdk #184: What does mapImage's data value represent?

nabe899, 2021-03-23:
I'm trying to get map information from Temi and use it.
When I logged the value of mapImage data, I found that the one-dimensional array contains values such as -1, 52, and 92. Does this value represent the probability of occupancy?
I expected the occupancy value to be 0 to 100 and -1 unknown.

DoubleWB, 2021-07-07:
This one dimensional array represents the greyscale values of a visual representation of the Temi's slam data that was used when its map was saved. If you multiply these values by 2.55 and cast them to integers, they can be passed into the alpha channel of a color and put into a bitmap of the size specified by the ```getRows()``` and ```getCols()``` functions on the ```MapImage```. (Kotlin examples of this process are present in the ```MapActivity``` in the Temi Sample App)

I could be wrong, but I don't think any of the underlying probabilistic data of the map or localization algorithm that the Temi uses is exposed.

zjn0505, 2021-07-08:
> This one dimensional array represents the greyscale values of a visual representation of the Temi's slam data that was used when its map was saved. If you multiply these values by 2.55 and cast them [...]

That's correct.

-1 is unknown place. 0-100 means how likely there is a lidar obstacle, so 0 is empty space, 100 is a solid wall.

Here is a python sample. You can do similar thing with the JSON data you got.

https://colab.research.google.com/drive/1ZJBSiaP46GU2qO5txedA77HjxfzP6jhc

nabe899, 2021-07-08:
I'm glad to receive an answer. Thank you so much.

closed 2021-07-08 by nabe899

Alex-Beh, 2021-11-03:
@nabe899 May I know how you log the mapImage information from Temi?

zjn0505, 2024-04-22:
This is a sample code where a simple map is drawn

https://github.com/robotemi/sdk/blob/6a4d03a62e7a075983770e883880eef836b7b90b/sample/src/main/java/com/robotemi/sdk/sample/MapActivity.kt#L167-L200