robotemi/sdk #186: .getLocations() returning an empty list

themissingCRAM, 2021-03-30:
I have some issues .getLocations, it returns an empty list for some reason even though in temicenter, I have explicitly entered my locations needed.  

[image]

Im not sure what I am doing wrong or if this is a bug. 
Hope someone could help.

lcgao, 2021-04-01:
It is too early to invoke APIs after getting the instance of `Robot` immediately, please try to invoke function `getLocations()` after robot is ready([listen to OnRobotReadyListener](https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.java#L253-L264)).

themissingCRAM, 2021-04-05:
Thanks that solved the problem

closed 2021-04-05 by themissingCRAM