robotemi/sdk #193: TEMI voice

kilmhd, 2021-04-28:
Hi, 
There is the possibility to change TEMI voice, directly via the SDK?

best

lcgao, 2021-04-30:
Currently no.

DanLechuga, 2022-03-05:
Unless it overwrites the OS, adding the voice packs, I tried it on my temi and it worked.

prashantsmp, 2022-11-09:
Can you guide me to **overwrites the OS, adding the voice packs**? @DanLechuga

zjn0505, 2022-11-09:
By changing temi voice, if you mean the change default voice engine for voice interaction, e.g. if you want temi to talk by the voice powered by AWS polly. Then you have override temi's default TTS engine.

Docs can be found here https://github.com/robotemi/sdk/wiki/Speech#setttsservice.

By doing so, whenever temi wants to speak something, like broadcasting weather, answering your request to save a location,the temi launcher app will send the speaking intents to your app through SDK, and wait for your TTS engine to handle them and send events back to temi launcher.

Currently temi TTS is powered by Google and Azure in multiple languages and voices, if you just want to speak in another language other than current system language, then it is already possible through SDK Robot.speak() function.